JamesCJ60/Universal-x86-Tuning-Utility #223: [FEATURE REQUEST] Undervolt missing in some CPUs despite enabled support

ShyVortex, 2024-07-29:
Hi everyone, I currently own the HP Pavilion Gaming 16 a-0046nl laptop, and as some of you may know, by default it is fully locked, meaning you cannot access any CPU tweaks or customize any power-related options.
That's why I followed [this guide](https://www.reddit.com/r/Hewlett_Packard/comments/o1zxuv/howto_unlock_undervolting_change_timings_on_hp/) when I got the laptop to unlock undervolting etc. on my CPU, which is the Intel Core i7-10750H.
Before trying out UXTU for a few weeks, I've been using ThrottleStop for years, and after following said guide every option was enabled, tweakable and working, including the possibility to undervolt the CPU which is unfortunately missing for some reason, even though I know the option actually exists for other CPUs.
I love the idea of having multiple custom presets, tweaking my dGPU directly from UXTU, and even the possibility of enabling a universal upscaler in games... but I miss being able to undervolt! It is especially needed in summer due to the high temperatures, so I hope support for this and eventually other unsupported CPUs can be added.

[image]

JamesCJ60, 2024-08-06:
Hi, I hope to add FIVR voltage control for supported Intel CPUs in the next release.

ShyVortex, 2024-08-09:
Thank you so much!

JamesCJ60, 2024-08-14:
This should now be working as of the latest build.

closed 2024-08-14 by JamesCJ60