JamesCJ60/Universal-x86-Tuning-Utility #224: Not a BUG but is the AMD Ryzen 7 7840HS with Radeon 780M graphics supported ?

JamVzxc, 2024-07-29:
Does Universal-x86-Tuning-Utility work on locked CPU's ? I tried creating a custom profile , and when I used HW monitor to check if the changes were applied nothing changed. So im not sure if the CPU cant be tweaked or is it the program ?

Regards
JamVzxc

Paw565pl, 2024-08-04:
Yes, it works like a charm for me.
[image]

JamVzxc, 2024-08-04:
@Paw565pl Do custom profiles work for you also ? I havent tried the presets but ill see if they change anything via HW monitor

Paw565pl, 2024-08-05:
@JamVzxc They work, but I have to apply them manually. After restart of the program only the premade preset is applied, but unfortunetly not my custom one 😞