JamesCJ60/Universal-x86-Tuning-Utility #225: [BUG] A wrong command line within premade preset "Performance"

mishuha, 2024-07-30:
The program has 4 command lines for presets:

Eco: 
--tctl-temp=95 --ppt-limit=65000 --edc-limit=90000 --tdc-limit=90000 

Balanced: 
--tctl-temp=95 --ppt-limit=95000 --edc-limit=130000 --tdc-limit=130000 

Performance: 
--tctl-temp=95 --ppt-limit=125000 --edc-limit=142 --tdc-limit=142000 

Extreme: 
--tctl-temp=95 --ppt-limit=170000 --edc-limit=230000 --tdc-limit=230000 

And in Performance "--edc-limit=**142**" instead of "--edc-limit=**142000**" seems like as a typo that leads to actual performance loose after activation.

CPU: Ryzen 9 5900X
[image]

Reproducible on program version 2.2.5.0
[image]

**Desktop:**
OS: Windows 10 Pro 22H2

JamesCJ60, 2024-08-06:
Thank you for bringing this to my attention. I will get this fixed in the next release.

JamesCJ60, 2024-08-14:
This should be fixed as of the latest build.

closed 2024-08-14 by JamesCJ60

mishuha, 2024-10-09:
Not fixed in V2.3.2 stable.
There are no presets at all.  UXTU line commands are empty for all presets.
[image]