JamesCJ60/Universal-x86-Tuning-Utility #229: [BUG] UXTU uses 10% of CPU at low load

realQuagsire, 2024-08-13:
**Describe the bug**
Launching UXTU utilizes 10% of my Ryzen 7840HS at low load and keeps it running close to the base clock speed of 3800 MHz.

**To Reproduce**
Steps to reproduce the behaviour:
1. Open UXTU
2. Open task manager
3. Look at CPU usage for UXTU

**Screenshots**
[image]
[image]

**Laptop (please complete the following information):**
 - OS Version: Windows 11 23H2
 - UXTU Version: 2.2.15.0
 - CPU: Ryzen 7 7840HS
 - Laptop: Dell G5 5535

**Additional Context**
I only use the basic adaptive mode settings and set the max CPU temperature to 80 degrees C and the max power limit to 54 watts. The only programs I had open were Microsoft Edge, Steam, and Sticky Notes.

realQuagsire, 2024-08-13:
I just noticed that if you turn off adaptive mode, the CPU usage returns to normal. Is it necessary for adaptive mode to use 10% of the CPU?

closed 2024-08-14 by realQuagsire

Splarkszter, 2026-03-29:
Did you solve the issue? I have the same behavior.