JamesCJ60/Universal-x86-Tuning-Utility #231: [BUG] After updating universal x86 tuning utility the application opens for a split second and immidiatly closes

M8M24, 2024-08-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS Version: [e.g. Windows 11 22H2]

**Additional context**
Add any other context about the problem here.

M8M24, 2024-08-18:
I have tried uninstalling and downloading it several times but the issue keeps reoccurring, i have tried different versions as well with the same result plus anything before version 2.3.0 Beta 1, says that their is a newer version available on my laptop, so i tried deleting any trace of it i could find and trying again but that didn't work, i also tried the remove universal x86 tuning utility option given by Universal.x86.Tuning.Utility.V2 i also tried the repair option but that didn't work either

muthu-25, 2024-08-31:
I am facing the same issue on unmodified Windows 11 Home. My laptop has Ryzen 5 3500U

rabiddoughnuts, 2024-09-18:
I am having the same issues on windows 11 home ryzen 7940hs. Until this latest update it worked and I was using it to undervolt

hamixd, 2024-10-09:
same here. 3500u. latest windows. 2.0.0 runs fine though

hamixd, 2024-10-09:
Ok guys I found the solution to this problem.

You all have to update to the latest chipset and graphics drivers for the software to work.

rabiddoughnuts, 2024-10-09:
> Ok guys I found the solution to this problem.
> 
> You all have to update to the latest chipset and graphics drivers for the software to work.

That was it, I have a framework, so the new drivers weren't available till the 2nd, but it works now, thanks. Shoulda known, it's always drivers.

muthu-25, 2024-10-12:
> Ok guys I found the solution to this problem.
> 
> You all have to update to the latest chipset and graphics drivers for the software to work.

Yup that did the trick, whilst my device was up to date, I needed to install drivers directly from AMD which I presume is still newer as manufacturer did not provide that.

JamesCJ60, 2026-02-21:
Hi, please ensure that you are using the latest drivers from your hardware vendors and that all the software prerequisites have been installed.

closed 2026-02-21 by JamesCJ60