JamesCJ60/Universal-x86-Tuning-Utility #233: [BUG] AMD Per-Core Curve Optimiser - changing value for CCD1 also changes it for CCD2

JonasDockx, 2024-08-20:
When I try to set my CO values per-core, I run into a problem. Changing a value in CCD1 sets the same value for the same number of core in CCD2. Say, for example, if I set Core 0 in CCD1 to -18 and I set Core 0 in CCD2 to -20, when saving, the CCD2 Core 0 gets overwritten to -18 too.

Steps to reproduce the behaviour:
1. Go to Custom Presets
2. Set values in CCD1 to the desired values
3. Set values in CCD2 to the desired values
4. Set a New Preset Name
5. Click Save current settings as new preset

I would expect this to save the values in CCD2 the way I set them.

CCD1 values I set:
[image]

CCD2 values I set before clicking save:
[image]

CCD2 values after clicking save and applying that preset:
[image]

 - OS Version: Windows 11 22H2

Andowa, 2024-09-10:
I encountered the same problem, although there is support for ccd2, its program seems to think that the parameters of ccd1 and ccd2 should be the same. I can adjust the settings of ccd2 separately, but when I save and read the file I saved again, the parameters in ccd1 and ccd2 are the same.

JamesCJ60, 2024-09-21:
This issue should now be fixed as of the latest build. Thanks for reporting the issue.

closed 2024-09-21 by JamesCJ60