JamesCJ60/Universal-x86-Tuning-Utility #236: [BUG] Partition table corruption after a few hours

bbanho, 2024-09-01:
**Describe the bug**
A clear and concise description of what the bug is.

Systematic Windows drive partition table corruption when running the software for about 6 hours. Could not find any dump or log in general. Many EFI and WinRE partitions files remained broken when trying to rewrite the p table backup. My SSD is a suspicious generic cheap drive. But the problem had been gone when I gave up using the software.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

_**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS Version: [e.g. Windows 11 22H2]_

**Additional context**
Add any other context about the problem here.

Don't hesitate asking me more info.

Computer:      LENOVO IdeaPad 3 15ALC6
CPU:           AMD Ryzen 5 5500U (Lucienne, LCN-A1)
               2100 MHz (21.00x100.0) @ 4042 MHz (40.50x99.8)
Motherboard:   LENOVO LNVNB161216
BIOS:          GLCN52WW, 12/09/2022
Chipset:       AMD Promontory/Bixby FCH
Memory:        8192 MBytes @ 1597 MHz, 22-22-22-52
               - 4096 MB PC25600 DDR4 SDRAM - Smart Modular SMS4WEC3C0K0446SCG
Graphics:      AMD Lucienne - Internal GPU [Lenovo]
               AMD Radeon Vega, 2048 MB DDR4 SDRAM
Drive:         WDC WD10SPZX-75Z10T1, 976.8 GB, Serial ATA 6Gb/s @ 6Gb/s
Drive:         Patriot M.2 P300 128GB, 125.0 GB, NVMe
Sound:         ATI/AMD Renoir/Cezanne - Display HD Audio Controller
Sound:         AMD Zen - Audio Processor - HD Audio Controller
Network:       RealTek Semiconductor RTL8822CE Wireless LAN 802.11ac PCI-E NIC
OS:            Microsoft Windows 11 Professional (x64) Build 22631.4112 (23H2)

bbanho, 2024-09-01:
Hardware failure under sustained load, the utility was just boosting it into a quicker crash. I'm feeling into using it to underclock lol

appreciate the tool, pretty useful no matter watt xD

shadowsofthesun, 2024-09-16:
This should be closed since the program is just exposing an underlying hardware failure?

bbanho, 2024-09-19:
Made more tests and has to be.  Hardware issues. Thank you for attention.

closed 2024-09-19 by bbanho