JamesCJ60/Universal-x86-Tuning-Utility #242: [FEATURE REQUEST]Please support Linux.

arabcian, 2024-09-14:
Tried to build PR you did on ryzenadj with experimental 7045hx series support with no luck it just hangs for some time and then errors out. I have a 7845hx and want to tweak it under Linux is there any possibility of Linux support in the future?

Vistaus, 2024-10-09:
I would also like to see that as I am a Linux user.

rabiddoughnuts, 2024-10-11:
I would also love to see this, ryzenadj or amdctl both "work" but this tool is so much easier and intuitive to use imo

Alastor-Ripley, 2024-11-26:
I would also love to see a linux version

JamesCJ60, 2025-01-03:
Hi all, I cannot get into much detail right now, but UXTU’s successor, which I have started development work on, aims to support Linux (provided nothing goes wrong) and will merge a few of my projects into one (e.g., UXTU, UXTU Handheld, Framework Hub, etc.).

arabcian, 2025-01-03:
Woah thats awesome to hear.

officiallymarky, 2025-06-04:
I'd love to see this on Linux, or an update on your new version.

JamesCJ60, 2025-07-29:
Hi all! Just wanted to keep you all in the loop. @loyvsc is currently working on a Linux port of this version of UXTU. It’s a work in progress, so we appreciate your patience and support. Again, big thanks to @loyvsc for taking this on!

arabcian, 2025-08-16:
Hello Guys Good news that JamesCJ's pull request on ryzenadj about DragonRange and FireRange preliminary support has accepted, now i can use curve optimizer on my 7845hx

Danielli64, 2025-09-02:
Yes!! Just tested RyzenAdj and set the Curve Optimizer to -25 on my 7945HX in Linux (Debian 13) and it bl..dy worked. :) Ran passmark and performance numbers i have never seen before. Nice!!!!

Danielli64, 2025-09-04:
> Hello Guys Good news that JamesCJ's pull request on ryzenadj about DragonRange and FireRange preliminary support has accepted, now i can use curve optimizer on my 7845hx

Thanks for the heads up. I would never have seen otherwise. It would be nice if they could add min and max CPU frequency. One core stops att -25 CO all others can run att -30 or less. Reducing the CPU frequency from 5450 to 5400 MHz the last core can also goes to -30. Even better, set frequency per core :) Tested this on Windows with SMUDebugTool where it is possible to set global frequency but not per core. You always want more more feature. :)