JamesCJ60/Universal-x86-Tuning-Utility #249: [BUG] Curve optimiser iGPU offset has no effect on 8945hs

andriy-kudrya, 2024-10-06:
**Describe the bug**
It seems that curve optimiser iGPU offset has no effect on 8945hs.
I've tried increasing and decreasing it but this doesn't affect iGPU power draw (verified with a smart plug and HWiNFO).

**Desktop (please complete the following information):**
 - OS Version: Windows 11 24H2
 - UXTU Version: 2.3.2
 - Hardware: Minisforum um890 pro

**Additional context**
Both _iGPU clock_ and _CPU curve optimiser_ work well.

andriy-kudrya, 2024-11-17:
Small update. The same behavior is observed on desktop Ryzen 8700g APU.

ObnubiladO, 2025-11-20:
It would be nice to know why it doesn't work, I have the same APU.

andriy-kudrya, 2026-03-01:
> as for 8945hs, duplicate of [#258](https://github.com/JamesCJ60/Universal-x86-Tuning-Utility/issues/258) as for 8700g, i am unsure...

It is not. 8845hs from linked issue doesn't support CO at all.