JamesCJ60/Universal-x86-Tuning-Utility #258: r7-8845hs

Dllzzz, 2024-11-07:
I used this software to undervolt my laptop equipped with an R7-8845HS, but I couldn't confirm if it took effect. Can you tell me if the 8845HS is compatible with this software?

neoskynet, 2024-12-07:
It also doesn’t work for me for r7-8845HS on an acer nitro 16 an16-42-r287 laptop. Does anyone know how to reduce the voltage on this processor?
maybe there is other software?

X4ero26, 2025-08-24:
> He pasado dias tratando de limitar el voltaje y consumo del R7-8845HS ya que en laptop lo maximo que deberia consumir es 45 w pero a mi me consume 55w y su voltaje es alto y la temperatura me ha llegado a los 100 grados, no encuentro ninguna solucion, ni con BIOS ya que mi laptop es HP y HP bloquea estas funciones y ningun software de AMD o de terceros es compatible con este procesador

Encontraste una solución? Me pasa algo parecido

closed 2026-02-21 by JamesCJ60

Sage222, 2026-04-28:
yeah i have the same CPU and from my observations of voltage under load with the latest version 26.2.0b4 the Curve Optimiser doesnt do anything.   Havent tried the iGPU yet..