JamesCJ60/Universal-x86-Tuning-Utility #262: Cannot install Net 7.0 since reached EOL.

jsalaz-Gramps, 2024-11-24:
**Describe the bug**

The installation of the installer, asks for .net 7.0, but this entry is out of date, and is no longer supported.

**To Reproduce**
Steps to reproduce the behaviour:
1. Attempt to install
2. When program asks to download .Net 7, it fails.
3. Program exits.

**Expected behavior**

Allow computer to install the .Net 9 platform, and continue installation.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 
OS: Windows 11 24H2

dartraiden, 2025-01-03:
Also, it's weird that the installer first requires .NET 7.0, and then the application requires 8.0

IAmEchino, 2025-02-06:
Yeah something's definitely bugged, I installed 8.0 (the version that the changelog says is compatible) and it still is asking for 7.0.

jsalaz-Gramps, 2025-02-06:
Let's just hope it is upgraded, because of the fast pushout of .Net 7.0.
It seems like it was rushed.  >nert 8.0 is better anyway.

LeGraon, 2025-02-20:
Hello. I have the same problem with UXTU V2.4.2 on Windows 11 24H2. The installer is still asking for .NET 7.

jsalaz-Gramps, 2025-02-20:
I figure that the project is dead, since there has been no response to our
concerns for 6-8 months.

LeGraon, 2025-03-02:
The project is not dead since there was an update to the version 2.4.5 one hour ago. New version but same issue...

jsalaz-Gramps, 2025-03-02:
OK!

LeGraon, 2025-03-05:
Actually I first installed .Net 7 then .Net 8 and it works well.