JamesCJ60/Universal-x86-Tuning-Utility #263: [BUG] Driver HwRwDrv.x64 "A Driver cannot load on this device"

xSimone, 2024-11-25:
Hello, some reason when launching the tool, I get this error.

[image]

My Windows installation is new, and I installed .net 7.

[image]

System Specs:
Asus Scar 18R 2024 model Laptop
i9 47900HX
RTX 4080m
64GB Crucial

muhalinzalma, 2024-12-16:
same i have winver 24H2 too and same problem i still search to resolve it. do u resolve the problem?

guilamu, 2025-02-22:
Just to confirm I have the exact same bug on any PC with Windows 11. The app is not functional anymore.

taublast, 2025-05-23:
Any news on this?

InanimateCrbnRod, 2025-06-21:
any updates? was excited to try this but I keep getting this message

aaron-manning, 2025-07-14:
Disable these settings in Windows 11

[image]

mu-coder69, 2025-08-06:
Same problem here. It didn't solve the issue.

LispEngineer, 2025-08-07:
This setting from @aaron-manning used to work to solve the problem above. However, something must have changed in Windows 11, because now I am getting the `HwRwDrv.x64` error again. 

I have Windows 11 Pro 24H2 build 26100.4351. Any ideas?