JamesCJ60/Universal-x86-Tuning-Utility #264: Random 400mhz when close to set Temp Limit

Brew-er, 2024-11-28:
**Describe the bug**
When setting a temp limit whether it be 90c or 75c, the CPU clocks plummet to 400mhz for awhile when it reaches/passes the temp limit set and clocks comes back up soon with lower temps. Even though without temp limit, it doesn't usually throttle around those temps. This doesn't seem to happen with just the Skin Temperature Limit active which is set to 45.

**To Reproduce**
Steps to reproduce the behaviour:
1. Tick Temperature Limit and set desired limit.
2. Apply.
3. Use the new preset in some CPU demanding games like Apex Legends.
4. Should throttle soon in a few minutes to 400mhz time to time.

**Expected behaviour**
To throttle by lowering the frequency just a bit below where it averaged during use to maintain close or below temp limit set without hurting performance drastically.

**Screenshots**
Note that it does hit lower than 400mhz but that might just be the same issue and 400/399mhz was a more common frequency that happened during stress.
Center values show the minimum frequency it clocked to during gameplay using HWMonitor
[image]

**Laptop**
- Acer Nitro 5 AN515-43-R3TY
- CPU: Ryzen 3550H 
- GPU: RX 560X
- OS Version: Windows 11 Pro 10.0.26100 Build 26100

**Additional context**
It drops further than 400mhz sometimes.
Game is also capped to 60 fps to further lower frame drops.
GPU is also undervolted a bit and is stable.

KuroSeinenbutV2, 2024-12-30:
i facing this issue too, sometime my Ryzen 7 7840HS running at 400mhz while playing Far cry 6, even my CPU temp are 60c

tando999, 2025-01-11:
anyone found fix yet??

DRW-22, 2025-02-15:
I have the same issue but only when using my Acer Nitro 5 with Ryzen 5 4600H. 
When I use my sister's ASUS TUF GAMING with the same processor, this problem does not occur.
So maybe this is somehow related to BIOS settings or something like this?

naff1s, 2025-03-06:
> I have the same issue but only when using my Acer Nitro 5 with Ryzen 5 4600H. When I use my sister's ASUS TUF GAMING with the same processor, this problem does not occur. So maybe this is somehow related to BIOS settings or something like this?
no this is definitely uxtu issue since ive also had same issue even on the new motherboard.
and deleting uxtu seem to fix the issue, although my older motherboard says otherwise....

and stay away from premade profile.... it mess things up and the dev should definitely have some apply button so the user dont accidentally click it and change their tdc and edc limit.... and the dev should be using only power limit on their premade profile and nothing else since not every mobo have the same tdc,edc , etc etc on their device