JamesCJ60/Universal-x86-Tuning-Utility #265: [FEATURE REQUEST] BD Prochot disable

adelage2011, 2024-12-08:
Is there any chance to add an option to disable BD PROCHOT like in Throttlestop? There is an increasing number of Ryzen laptop that face the Prochot EXT flag, which renders the devices almost unusable due to ridiculously low cpu frequency (in my case 5900hx@0.4ghz). In most cases temps are perfectly fine, and most likely there is some hidden sensor that is failing, reporting a false issue that triggers BD Prochot, and people manage to sort it out by disabling it with Throttlestop (Intel) or Ryzen Master (desktop), neither of which work with Ryzen mobile cpu.

KuroSeinenbutV2, 2024-12-30:
i facing this issue too, sometime my Ryzen 7 7840HS running at 400mhz while playing Far cry 6, even my CPU temp are 60c

rucascampeoum, 2025-02-17:
I'm having this problem too and I'm looking for a way to disable BD PROCHOT on my Lenovo Legion 5. Is this feature still not available?

lelexy100, 2025-02-20:
I'm also in dire need of a BD PROCHOST disable. My laptop works great while connected to the original charger, but I get the damned flag on battery or on a spare charger. Any pointers to how it's implemented in Ryzen Master or ThrottleStop? I can't use any of those since I'm on AMD and a laptop :( . I'd gladly try to make a PR if I had any idea on how this disable works.

MASTERGUY, 2025-05-16:
+1 my laptop also has same issue i am unable to use it & i am not able to disable BD-PROCHOT because my CPU doesnt support ryzen master 🙁 please help is appreciated

JamesCJ60, 2025-05-16:
PROCHOT cannot be disbaled on Ryzen APUs/CPUs that do not support overclocking.

closed 2025-05-16 by JamesCJ60