JamesCJ60/Universal-x86-Tuning-Utility #272: [BUG] UXTU refuses to open

ImCoolio, 2025-01-16:
Title. Attempting to open shows it in task manager for 3-5 seconds and disappears.

Windows 11, 24H2, Ryzen 7 5800H on a Lenovo Legion 5 15ACH6.

loyvsc, 2025-02-10:
Try install last drivers.

rmarmitt, 2025-03-04:
Same here. Just tested with 3 different versions. The software open and stay opened for like 30s them just close itself without any error. I can see the processor name in the main frame. How can I provide some logs? This maybe help in some way.

DuCun, 2025-03-12:
> Same here. Just tested with 3 different versions. The software open and stay opened for like 30s them just close itself without any error. I can see the processor name in the main frame. How can I provide some logs? This maybe help in some way.
> 
> Windows 11 23H2, Intel Core i5-9300HF - Lenovo L340 15IRH

`C:\Windows\System32\Logs` -> `uxtu_log********.txt`

weedy, 2025-04-23:
The problem is #220