JamesCJ60/Universal-x86-Tuning-Utility #295: [BUG]

Byteme-dot, 2025-04-08:
App not launching.

I installed the latest version v2.4 which was released 2 days ago. I double click on it to launch and it doesn't.

When i double click it, it appears in my system tray for few seconds and then disappears

OS: Windows 11 24H2 (OS BUILD 26100.3775)

nopQone, 2025-04-20:
U need instal FULL SDK 8.0, uxtu instaler provide link only to runtime but not workin and uxtu crashing after openin.

Byteme-dot, 2025-04-21:
I installed both SDK 8.0.... and ASP NET CORE RUNTIME 8.0..... from here https://dotnet.microsoft.com/en-us/download/dotnet/8.0

It still doesnt work.

nopQone, 2025-04-21:
uh, actually: 22h2, .NET SDK 7.0.410, 8.0.408, 9.0.203, vc++ 05, 08, 10, 13, 15-22.. and workin.

weedy, 2025-04-23:
The problem is #220

Bukollah, 2025-05-29:
I have the same issue. Since I am not a computer science major I didn't understand any of these. I have download the "Universal x86 Tuning Utility V2.5.1". Now what do I need to make it working properly.