JamesCJ60/Universal-x86-Tuning-Utility #296: [BUG] 7640HS reapply issue

monklaff, 2025-04-17:
I have my laptop with 7640HS processor and everything works fine while i apply presets manually, but they are not active after system reboot. Settings like ‘auto reapply’ and ‘reapply on startup’ do not work.

Truth-77, 2025-04-21:
Been having this issue for months as well. It's not just the reapply every X seconds option either, the reapply upon plugging in isn't working either. Actually, some aspects of it are working : Refresh rate. The power limit settings are not applying though.

Chichi168188, 2025-04-29:
Z13 AIMax 395. Everything is working manually , but couldn't have the auto reapply after rebooting.

monklaff, 2025-05-05:
Well, after the latest update everything works fine now! Thank you very much!

closed 2025-05-05 by monklaff

wmd0701, 2025-05-05:
> Z13 AIMax 395. Everything is working manually , but couldn't have the auto reapply after rebooting.

dude how much offset do you undervolt? I also use uxtu on z13 2025. Everytime when I undervolt more than -30 no audio device cannot be detected.