JamesCJ60/Universal-x86-Tuning-Utility #301: [BUG] No audio device after auto reapply - ASUS FLOW Z13 2025

Marruda9, 2025-05-13:
**Describe the bug**
Enabling "Auto Reapply" breaks sound driver.  "High Def Audio Controller" in device manager

**To Reproduce**
Uncheck auto reapply. 
Uninstall HD audio controller driver.
Scan devices to reapply driver
Checkmark auto reapply in UXTU for 3 sec
At update volume control shows no audio device

**Desktop (please complete the following information):**
 - OS Version: Windows 11 24H2

**Additional context**
Asus flow z13 2025.  Ryzen AI max+ 395. Running G helper.  No Asus services.  Armoury Crate uninstalled.

Aleksis77, 2025-06-28:
Yes, everything is true - this is preserved from version to version. Therefore, I try not to enter the settings once again.)) So that the counter does not drop 3 seconds by default. It is probably difficult to fix it - if the developer is knowing about it, leaves everything as it is.

Ajanni2017, 2025-09-19:
> Yes, everything is true - this is preserved from version to version. Therefore, I try not to enter the settings once again.)) So that the counter does not drop 3 seconds by default. It is probably difficult to fix it - if the developer is knowing about it, leaves everything as it is.

Im having the exact same issue on my zephyrus g16 2024 AMD HX 370 model. The sound driver crashes when i set auto reapply to on. So everytime my laptop wakes up from sleep/hibernation i have to reapply the undervolt